pyros-dev/catkin_pip #26: not able to import classes when using catkin_pip to make pure python package

dhirajdhule, 2016-07-20:
I am integrating a pure python module(pure python APIs as a pyros_client) into catkin workspace. I followed your instructions in readme and edited cmakelist.txt accordingly, also edited setup.py as a python package and package.xml as well to use catkin_pip to build tool dependency.
My package structure is as following
src
      -----python_package_name
            __init__.py
            api_using_pyros_client_classes_that_I_need_to_import.py
      setup.py
      CMakeLists.txt
      package.xml

After usual catkin_make and sourcing I am able to import the package, however I am not able to import the classes from other file. Not even any objects inside __init__.py. I am getting attribute error. 

Is there anything else that I need to specify?

I have used cmakelists.txt, setup.py, package.xml from pyros source and made changes in them accordingly.

dhirajdhule, 2016-07-20:
I am not planning on building pip package though. Users will be importing libraries inside this package as APIs. So you can suggest me if I really need to use catkin_pip? Is using catkin_python a better way to catkinize pure python code if I dont plan on packaging it as pip package or use features like installing dependencies using pip, etc.

asmodehn, 2016-07-20:
If you're integrating a pure python package, and if all dependencies are from source, then normal catkin python way (catkin_python_setup, etc.) will work.
However you won't be able to rely on pip packages (only on old catkin packages) and you cannot release your python package on pip ( only as deb, after catkinizing **all** dependencies )

Note : I am still unsure about how I could (or why I would) package as debs with catkin_pip, this work is still ongoing. But packaging as pip is trivial (as easy a normal python pip packaging at least).

Latest catkin_pip released version should work fine (not latest source - indigo-devel is currently mysteriously broken).

Did you try to do these imports directly using only python (no catkin magic) ? Just in case it is a problem of python package hierarchy, configuration, and imports (can be tricky to get right sometimes).

Since you can import the package I would assume catkin_pip setting works fine ( only sets PYTHONPATH), the rest depends on python itself and the folder structure /vs/ import behavior...

In any case, here is a simple pure python project, using catkin_pip, that you can use as an example : https://github.com/asmodehn/pyros-config

Hope this helps, Let me know if you manage to gather more details.

asmodehn, 2016-08-12:
I recently spend a bit of time writing and structuring the catkin_pip documentation. Please have a look in the indigo-devel branch, and let me know what you think about it.

There is still lots of details to add, but I think it covers all the intended use cases.

Feel free to PR any modifications you would like to make to these docs.
Note however that regarding the behavior of `import` from a python environment, this should be addressed in the pyros-setup documentation (still a work in progress)

Thanks !

asmodehn, 2017-03-14:
No activity for a long time -> Problem has likely be solved -> closing.

closed 2017-03-14 by asmodehn