pyros-dev/catkin_pip #55: OSError with catkin_make

naveedhd, 2016-12-01:
```
-- +++ processing catkin package: 'pyros_config'
-- ==> add_subdirectory(pyros_config)
-- Loading catkin-pip.cmake from /opt/groot/bootstrap_ws/devel/share/catkin_pip/cmake... 
-- Loading catkin-pip-requirements.cmake from /opt/groot/bootstrap_ws/devel/share/catkin_pip/cmake... 
-- Loading catkin-pip-prefix.cmake from /opt/groot/bootstrap_ws/devel/share/catkin_pip/cmake... 
-- Loading catkin-pip-env.cmake from /opt/groot/bootstrap_ws/devel/share/catkin_pip/cmake... 
-- Loading catkin-pip-runcmd.cmake from /opt/groot/bootstrap_ws/devel/share/catkin_pip/cmake... 
-- Loading catkin-pip-setup.cmake from /opt/groot/bootstrap_ws/devel/share/catkin_pip/cmake... 
-- Loading catkin-pip-env.cmake from /opt/groot/bootstrap_ws/devel/share/catkin_pip/cmake... 
-- Loading catkin-pip-runcmd.cmake from /opt/groot/bootstrap_ws/devel/share/catkin_pip/cmake... 
-- Loading catkin-pip-prefix.cmake from /opt/groot/bootstrap_ws/devel/share/catkin_pip/cmake... 
CMake Warning (dev) at /opt/groot/bootstrap_ws/devel/share/catkin_pip/cmake/catkin-pip-env.cmake:27 (message):
  Incomplete catkin-pip setup detected.This is expected if you use catkin-pip
  from source, and did not source the current develspace yet.Quick patch will
  be applied to the current cmake process environment.
Call Stack (most recent call first):
  /opt/groot/bootstrap_ws/devel/share/catkin_pip/cmake/catkin-pip-prefix.cmake:127 (catkin_pip_check_env)
  /opt/groot/bootstrap_ws/devel/share/catkin_pip/cmake/catkin-pip.cmake:63 (catkin_pip_setup_prefix)
  /opt/groot/bootstrap_ws/devel/share/catkin_pip/cmake/catkin_pipConfig.cmake:190 (include)
  /opt/ros/indigo/share/catkin/cmake/catkinConfig.cmake:76 (find_package)
  pyros_config/CMakeLists.txt:11 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Adding /opt/groot/bootstrap_ws/build/catkin_pip_env/lib/python2.7/site-packages to PYTHONPATH
-- Adding /opt/groot/bootstrap_ws/build/catkin_pip_env/bin to PATH
-- Catkin pip PYTHONPATH : /opt/groot/bootstrap_ws/build/catkin_pip_env/lib/python2.7/site-packages:/opt/groot/bootstrap_ws/devel/lib/python2.7/site-packages:/opt/groot/bootstrap_ws/devel/lib/python2.7/dist-packages:/opt/groot/messages_ws/devel/lib/python2.7/dist-packages:/opt/yujin/amd64/indigo-devel/lib/python2.7/dist-packages:/opt/ros/indigo/lib/python2.7/dist-packages:/opt/groot/bootstrap_ws/src/pyros
-- Catkin pip PATH : /opt/groot/bootstrap_ws/build/catkin_pip_env/bin:/opt/groot/messages_ws/devel/bin:/opt/yujin/amd64/indigo-devel/bin:/opt/ros/indigo/bin:/usr/local/heroku/bin:/home/naveed/.nvm/versions/node/v7.0.0/bin:/home/naveed/.rvm/gems/ruby-2.2.4/bin:/home/naveed/.rvm/gems/ruby-2.2.4@global/bin:/home/naveed/.rvm/rubies/ruby-2.2.4/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/home/naveed/.rvm/bin:/home/naveed/.rvm/bin
--     ... Catkin pip was not found in /opt/groot/bootstrap_ws/build/catkin_pip_env/bin ...
--     System pip version detected : 1.5.4
--     System pip install detected options : 
--       --user
--     ... Retrieving catkin_pip requirements using system pip ...
--     ... Done ... [0]: 
--     ... Running python -m pip install --ignore-installed -r /opt/groot/bootstrap_ws/devel/share/catkin_pip/cmake/catkin-pip-fixups.req --src /opt/groot/bootstrap_ws/src --exists-action b --prefix /opt/groot/bootstrap_ws/build/catkin_pip_env ...
--     ... Done ... [0]: Collecting pip (from -r /opt/groot/bootstrap_ws/devel/share/catkin_pip/cmake/catkin-pip-base.req (line 3))
  Using cached pip-9.0.1-py2.py3-none-any.whl
Collecting setuptools (from -r /opt/groot/bootstrap_ws/devel/share/catkin_pip/cmake/catkin-pip-base.req (line 4))
  Using cached setuptools-29.0.1-py2.py3-none-any.whl
Collecting nose (from -r /opt/groot/bootstrap_ws/devel/share/catkin_pip/cmake/catkin-pip-fixups.req (line 7))
  Using cached nose-1.3.7-py2-none-any.whl
Collecting pytest (from -r /opt/groot/bootstrap_ws/devel/share/catkin_pip/cmake/catkin-pip-fixups.req (line 9))
  Using cached pytest-3.0.4-py2.py3-none-any.whl
Collecting pytest-cov (from -r /opt/groot/bootstrap_ws/devel/share/catkin_pip/cmake/catkin-pip-fixups.req (line 10))
  Using cached pytest_cov-2.4.0-py2.py3-none-any.whl
Collecting pytest-timeout (from -r /opt/groot/bootstrap_ws/devel/share/catkin_pip/cmake/catkin-pip-fixups.req (line 11))
  Using cached pytest_timeout-1.2.0-py2.py3-none-any.whl
Collecting py>=1.4.29 (from pytest->-r /opt/groot/bootstrap_ws/devel/share/catkin_pip/cmake/catkin-pip-fixups.req (line 9))
  Using cached py-1.4.31-py2.py3-none-any.whl
Collecting coverage>=3.7.1 (from pytest-cov->-r /opt/groot/bootstrap_ws/devel/share/catkin_pip/cmake/catkin-pip-fixups.req (line 10))
Installing collected packages: pip, setuptools, nose, py, pytest, coverage, pytest-cov, pytest-timeout
Successfully installed coverage-4.2 nose-1.3.7 pip-9.0.1 py-1.4.31 pytest-3.0.4 pytest-cov-2.4.0 pytest-timeout-1.2.0 setuptools-29.0.1

-- Found catkin_pip pip command at /opt/groot/bootstrap_ws/build/catkin_pip_env/bin/pip.
-- Loading catkin-pip-env.cmake from /opt/groot/bootstrap_ws/devel/share/catkin_pip/cmake... 
-- Loading catkin-pip-runcmd.cmake from /opt/groot/bootstrap_ws/devel/share/catkin_pip/cmake... 
-- Loading catkin-pip-setup.cmake from /opt/groot/bootstrap_ws/devel/share/catkin_pip/cmake... 
-- Loading catkin-pip-prefix.cmake from /opt/groot/bootstrap_ws/devel/share/catkin_pip/cmake... 
-- Loading catkin-pip-requirements.cmake from /opt/groot/bootstrap_ws/devel/share/catkin_pip/cmake... 
-- Catkin_pip using Python nosetests: /opt/groot/bootstrap_ws/build/catkin_pip_env/bin/nosetests-2.7
-- Catkin_pip using Python py.test: /opt/groot/bootstrap_ws/build/catkin_pip_env/bin/py.test
-- Catkin_pip using Python nosetests: /opt/groot/bootstrap_ws/build/catkin_pip_env/bin/nosetests-2.7
--     ... Running /opt/groot/bootstrap_ws/build/catkin_pip_env/bin/pip install -e /opt/groot/bootstrap_ws/src/pyros_config --no-dependencies --prefix /opt/groot/bootstrap_ws/devel ...
--     ... Done ... [2]: Obtaining file:///opt/groot/bootstrap_ws/src/pyros_config
Installing collected packages: pyros-config
  Found existing installation: pyros-config 0.1.5
    Uninstalling pyros-config-0.1.5:

-- Command /opt/groot/bootstrap_ws/build/catkin_pip_env/bin/pip install -e /opt/groot/bootstrap_ws/src/pyros_config --no-dependencies --prefix /opt/groot/bootstrap_ws/devel FAILED !
CMake Error at /opt/groot/bootstrap_ws/devel/share/catkin_pip/cmake/catkin-pip-runcmd.cmake:37 (message):
  Exception:

  Traceback (most recent call last):

    File "/opt/groot/bootstrap_ws/build/catkin_pip_env/lib/python2.7/site-packages/pip/basecommand.py", line 215, in main
      status = self.run(options, args)
    File "/opt/groot/bootstrap_ws/build/catkin_pip_env/lib/python2.7/site-packages/pip/commands/install.py", line 342, in run
      prefix=options.prefix_path,
    File "/opt/groot/bootstrap_ws/build/catkin_pip_env/lib/python2.7/site-packages/pip/req/req_set.py", line 778, in install
      requirement.uninstall(auto_confirm=True)
    File "/opt/groot/bootstrap_ws/build/catkin_pip_env/lib/python2.7/site-packages/pip/req/req_install.py", line 754, in uninstall
      paths_to_remove.remove(auto_confirm)
    File "/opt/groot/bootstrap_ws/build/catkin_pip_env/lib/python2.7/site-packages/pip/req/req_uninstall.py", line 115, in remove
      renames(path, new_path)
    File "/opt/groot/bootstrap_ws/build/catkin_pip_env/lib/python2.7/site-packages/pip/utils/__init__.py", line 267, in renames
      shutil.move(old, new)
    File "/usr/lib/python2.7/shutil.py", line 300, in move
      rmtree(src)
    File "/usr/lib/python2.7/shutil.py", line 252, in rmtree
      onerror(os.remove, fullname, sys.exc_info())
    File "/usr/lib/python2.7/shutil.py", line 250, in rmtree
      os.remove(fullname)

  OSError: [Errno 13] Permission denied:
  '/opt/ros/indigo/lib/python2.7/dist-packages/pyros_config/_version.py'

Call Stack (most recent call first):
  /opt/groot/bootstrap_ws/devel/share/catkin_pip/cmake/catkin-pip-setup.cmake:97 (catkin_pip_runcmd)
  pyros_config/CMakeLists.txt:16 (catkin_pip_setup)

-- Configuring incomplete, errors occurred!
See also "/opt/groot/bootstrap_ws/build/CMakeFiles/CMakeOutput.log".
See also "/opt/groot/bootstrap_ws/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed
```

asmodehn, 2016-12-01:
It looks like the install of pyros_config (from source) conflicted with a
system installed old version of the same package... maybe catkin_pip should
just use "--ignore-installed" in that case as well ?

asmodehn, 2016-12-06:
This might be the same issue as https://github.com/asmodehn/catkin_pip/issues/58
Temporary fix by pinning pip down to 8.1.2, until we can actually solve it...