pyros-dev/catkin_pip #94: Sourcing path points to jenkins

naveedhd, 2017-03-13:
In local `messages` workspace, the PYTHONPATH points to jenkin's `../path_prepend.sh`. 

While sourcing the workspace:

```
bash: /var/lib/jenkins/bootstrap/devel/devel/share/catkin_pip/cmake/scripts/path_prepend.sh: No such file or directory
bash: /var/lib/jenkins/bootstrap/devel/devel/share/catkin_pip/cmake/scripts/path_prepend.sh: No such file or directory
bash: /var/lib/jenkins/bootstrap/devel/devel/share/catkin_pip/cmake/scripts/path_prepend.sh: No such file or directory
bash: grep: No such file or directory
bash: awk: No such file or directory
```

naveedhd, 2017-03-13:
Out of [`gopher`, `bootstrap`, `desktop`, `messages`] this is only happening this messages:

```
/opt/groot/messages_ws/devel/env.sh: 19: /opt/yujin/amd64/indigo-devel/etc/catkin/profile.d/42.site_packages.sh: /var/lib/jenkins/bootstrap/devel/devel/share/catkin_pip/cmake/scripts/path_prepend.sh: not found
/opt/groot/messages_ws/devel/env.sh: 20: /opt/yujin/amd64/indigo-devel/etc/catkin/profile.d/42.site_packages.sh: /var/lib/jenkins/bootstrap/devel/devel/share/catkin_pip/cmake/scripts/path_prepend.sh: not found
/opt/groot/messages_ws/devel/env.sh: 23: /opt/yujin/amd64/indigo-devel/etc/catkin/profile.d/42.site_packages.sh: /var/lib/jenkins/bootstrap/devel/devel/share/catkin_pip/cmake/scripts/path_prepend.sh: not found
```

naveedhd, 2017-03-13:
The other workspaces have a local workspace in overlay while the messages have overlay with only binaries which could be related to this:

```
bootstrap_ws >  yujin_make
-- This workspace overlays: /opt/groot/messages_ws/devel;/opt/yujin/amd64/indigo-devel;/opt/ros/indigo
```

```
messages_ws >  yujin_make
-- This workspace overlays: /opt/yujin/amd64/indigo-devel;/opt/ros/indigo
```

asmodehn, 2017-03-13:
This looks like bash gets broken when your workspace uses scripts from an old catkin_pip version...
Did you make sure you build from scratch (removing devel and build folder) after updating catkin_pip ?
The new version is not compatible with the previous one...

naveedhd, 2017-03-13:
Yes, still happens with removing devel and build folders. (clearing these also then cause more workspaces to have this error cause of no local underlay ws)

naveedhd, 2017-03-13:
The crash error in case I am missing something or connecting two different things:

```
-- Using these message generators: gencpp;genlisp;genpy
/opt/ros/indigo/share/genmsg/cmake/pkg-genmsg.cmake.em:56: error: <type 'exceptions.ImportError'>: No module named genmsg
Traceback (most recent call last):
  File "/usr/bin/empy", line 3276, in <module>
    if __name__ == '__main__': main()
  File "/usr/bin/empy", line 3274, in main
    invoke(sys.argv[1:])
  File "/usr/bin/empy", line 3257, in invoke
    interpreter.wrap(interpreter.file, (file, name))
  File "/usr/bin/empy", line 2262, in wrap
    self.fail(e)
  File "/usr/bin/empy", line 2253, in wrap
    callable(*args)
  File "/usr/bin/empy", line 2326, in file
    self.safe(scanner, done, locals)
  File "/usr/bin/empy", line 2368, in safe
    self.parse(scanner, locals)
  File "/usr/bin/empy", line 2388, in parse
    token.run(self, locals)
  File "/usr/bin/empy", line 1403, in run
    interpreter.execute(self.code, locals)
  File "/usr/bin/empy", line 2565, in execute
    exec(statements, self.globals)
  File "<string>", line 5, in <module>
ImportError: No module named genmsg
CMake Error at /opt/ros/indigo/share/catkin/cmake/safe_execute_process.cmake:11 (message):
  
  execute_process(/opt/groot/messages_ws/build/catkin_generated/env_cached.sh
  "/usr/bin/python" "/usr/bin/empy" "--raw-errors" "-F"
  "/opt/groot/messages_ws/build/rocon_msgs/concert_workflow_engine_msgs/cmake/concert_workflow_engine_msgs-genmsg-context.py"
  "-o"
  "/opt/groot/messages_ws/build/rocon_msgs/concert_workflow_engine_msgs/cmake/concert_workflow_engine_msgs-genmsg.cmake"
  "/opt/ros/indigo/share/genmsg/cmake/pkg-genmsg.cmake.em") returned error
  code 1
Call Stack (most recent call first):
  /opt/ros/indigo/share/catkin/cmake/em_expand.cmake:25 (safe_execute_process)
  /opt/ros/indigo/share/genmsg/cmake/genmsg-extras.cmake:303 (em_expand)
  rocon_msgs/concert_workflow_engine_msgs/CMakeLists.txt:30 (generate_messages)

-- Configuring incomplete, errors occurred!
See also "/opt/groot/messages_ws/build/CMakeFiles/CMakeOutput.log".
See also "/opt/groot/messages_ws/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed
```

asmodehn, 2017-03-13:
I tested and I have the same issue, as soon as I install the version built on jenkins...
I think it s because of a bug when installing the package, so that breaks the jenkins build of it.
I will need to investigate...

In the mean time this should fix it for everyone : https://github.com/yujinrobot/yujin_tools/commit/04609b29eccc33ea8235c1837d5cd455c87ad5af

So updating your binaries tomorrow should solve it, until I find the problem and fix it...

asmodehn, 2017-03-14:
fix has been implemented in gopher-devel branch.
@naveedhd Please close this as soon as you have confirmed the bug doesn't occur anymore.

naveedhd, 2017-03-14:
tested with gopher-devel, the problem doesn't occur now.

closed 2017-03-14 by naveedhd