pyros-dev/catkin_pip #109: Installing catkin python package with pip dependencies as virtualenv

asmodehn, 2017-04-14:
This issue is for discussion and gathering information regarding how we can install a python package hierarchy as _one_ ROS package.

This would make deploying complex python framework on ROS much easier.

The target here is to be able to package the whole virtual environment as a ROS package, because OSRF deploys ROS software this way. When containers or other means of deploying ROS packages will emerge, we will adapt to it.

The best existing solution candidate I could find so far is [dh_virtualenv](http://dh-virtualenv.readthedocs.io/en/1.0/info.html)

@k-okada FYI

asmodehn, 2017-04-14:
It looks like it already happened for fpm : https://github.com/jordansissel/fpm/issues/697

asmodehn, 2017-04-14:
Note : we do not want to change our current `catkin_pip_package` for this, since it is already working and there are many packages released in ROS based on it.

But we might want another API, like `catkin_pip_package_virtualenv`, that would manage the install differently...

As far as I can see, we shouldn't need to change the way we handle devel space for this.

k-okada, 2018-02-08:
Hi @asmodehn 

sorry for long silence, I have chance to back to this issue. see 
https://github.com/k-okada/catkin_pip/commit/e82f66663c2132bf78bd1e8f2c7b19f11c0d74ab
for `catkin_pip_package_virtualenv` example

I am testing this with https://github.com/k-okada/gdown, using `catkin_make` and use python code from python executables within virtualenv.
``
$ rosrun gdown python2
Python 2.7.12 (default, Nov 20 2017, 18:23:56) 
[GCC 5.4.0 20160609] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import gdown
>>> gdown
<module 'gdown' from '/home/k-okada/catkin_ws/ws_catkin_pip/src/gdown/gdown/__init__.pyc'>
>>> import pyros_setup
>>> pyros_setup
<module 'pyros_setup' from '/home/k-okada/catkin_ws/ws_catkin_pip/devel/share/gdown/catkin_pip_env/lib/python2.7/site-packages/pyros_setup/__init__.pyc'>
```

I'm not sure how we can use `dh-virtualenv` because that is happens in 'dpkg-buildpackage` stage. Are you suggesting we should modify `python-bloom` to installing catkin python package with pip dependencies as virtualenv  ?

asmodehn, 2018-02-10:
@k-okada recently, I was made aware of https://github.com/locusrobotics/catkin_virtualenv

I haven't tried it yet. If you do please let me know what you think about it. I dont know yet how it manages package releases...

I am wondering which approach makes more sense, or if there is a reason to keep both catkin_pip and catkin_virtualenv around...

Keeping pip compatibility for different versions in catkin_pip can be a little troublesome, mostly because I am not building a proper virtualenv, so I am not sure if it is worth to keep doing things the "custom" way as it is currently...

k-okada, 2018-02-21:
@asmodehn Thanks for info, I tried `catkin_virtualenv` in small example and seems ok for me. We'll test in more complex example and let you know the result.

Cc: @yuyu2172