pyros-dev/catkin_pip #143: Pass 'extra args' to catkin_pip_package(..) on to catkin_package()?

gavanderhoorn, 2017-08-09:
Use-case: a Python package that also has a `CFG_EXTRAS` file to install.

`catkin_pip_package(..)` calls `catkin_package(..)` internally ([here](https://github.com/pyros-dev/catkin_pip/blob/6c0d7599c28e3497cb649ccccd8bff7b5e125159/cmake/catkin-pip-package.cmake.in#L121)), but it does not support passing any arguments to it.

For packages that are not ROS pkgs (yet), but do want to make use of the `catkin_package(.. CFG_EXTRAS ..)` functionality, there is currently no way to do that.

It would be nice if that could be supported.

asmodehn, 2017-08-10:
Actually another option is to change the API : 
`catkin_pip_package` -> `catkin_pip_target`, since we create an actual cmake target, it would make some sense... and let the user call `catkin_package` as usual with the usual options...
It would make the cmake code simpler (which is always good).

And for the time being we can add a bwcompat `catkin_pip_package` function that does call both. (I m planning to remove it after packages using it have migrated (probably for catkin_pip 0.3).

Any preference ?

gavanderhoorn, 2017-08-10:
My current use-case is for integrating Python-based linters that lint stuff that is not Python. Lots of those are written in Python, and converting them into a Catkin pkg is the easiest way to get them to work.

So my main interest is in `CFG_EXTRAS` at the moment.

> Actually another option is to change the API :
`catkin_pip_package` -> `catkin_pip_target`, since we create an actual cmake target, it would make some sense... and let the user call `catkin_package` as usual with the usual options...
It would make the cmake code simpler (which is always good).

yeah, that is sort of what I did (in a very hacky way: https://github.com/gavanderhoorn/catkin_pip/commit/947dd386f1261dfdc235d8a52622d671e0eca6da).

It seems to work and would address this without too much additional work.

asmodehn, 2017-08-10:
I just committed a new version in `devel` branch that fixes this. Please try it out, and if everything is fine I ll make a new release (0.2.3) soon for (indigo, kinetic, lunar).

gavanderhoorn, 2017-08-10:
I commented on #144.

Seems to work, but as I wrote there, really minimal testing.

asmodehn, 2017-08-11:
This package is quite well tested, so if all tests are passing, we re good to release ;-)
Thanks for the check !

asmodehn, 2017-08-11:
0.2.3 has been released with a fix for this. closing.

closed 2017-08-11 by asmodehn